vongomben/smart-foot-brace #5: Definire comportamenti ed usi "corretti" e "sbagliati" del tutore

vongomben, 2023-06-21:
Alla luce delle letture [1] e dei limite tecnici di questo sensore, abbiamo un dato a 12 bit [2] per ogni GRF. 

Il dato c'é, ma quali livelli di pressione sono corretti, e quali sbagliati? 

### Quale [euristica](https://it.wikipedia.org/wiki/Algoritmo_euristico) dobbiamo implementare per capire se il tutore viene usato non correttamente? 

[1]
### Various links

[ZNX-01 Sensing Mat,Foot Pressure Sensor Insole Sensor Plastic Sensing Mat Thin Film Pressure Sensor Thin Film Intelligent Insole Sensor Foot Pressure Sensor](https://www.amazon.com/ZNX-01-Sensing-Pressure-Plastic-Intelligent/dp/B09PF5TNS8)
[Gait Phase Recognition Using Fuzzy Logic Regulation with Multisensor Data Fusion](https://www.hindawi.com/journals/js/2021/8776059/)
[Foot Pronation Detection Based on Plantar Pressure Measurement](https://www.researchgate.net/publication/346509459_Foot_Pronation_Detection_Based_on_Plantar_Pressure_Measurement)

### Specification

Item Type Sensing Mat
Material Plastic
Model ZNX-01
Thickness Approx. 0.45mm / 0.02in
Single Point Range Approx. 10 kg / 22 lb
Response Point Range Approx. 400 g / 14.1 oz
Durability >1 million times
Initial Resistance >10MQ (no load)
Test Voltage Typical value DC3.3V
Response Time <1ms
Recovery Time <15ms
Working Temperature -20℃~60℃
Electromagnetic Interference No generation
Electrostatic Discharge ESD Not sensitive

[2] (Risoluzione dell'ADC di 12 bit: 212 = 4096 livelli di quantizzazione)

marioantonaci, 2023-06-21:
Al momento, per la consegna del 5 luglio non ci sarà bisogno di ragionare su valori limite e soglie. Dobbiamo dimostrare invece che il sensore restituisce valori di pressione, possibilmente graficandoli sul pc

vongomben, 2023-06-22:
Perfetto. E' già così