vongomben/smart-foot-brace #10: Arduino Nano 33 IoT problems in sending JSON

vongomben, 2023-06-26:
Hi! 
Apparently I'm facing an issue which is more interesting and hard than my capabilities. 
When using `Arduino_JSON` vers 0.2.0 and `Arduino_LSM6DS3` 1.0.2 I find an interesting issue in dealing with the IMU and the reading of 8 different analog readings. 

The code that "silences" the Serial port is this one. 

NOTE: if I comment the JSON incapsulation of the analog readings everything keeps on stopping (hence that is not the problem).
Apparently the problem is the 8 analog readings. 

Fun Fact: if I add one reading and one JSON incapluation at a time, everything "breaks" when I reach two readings printed (alongside the JSON

Any help from @facchinm

```
#include <Arduino_LSM6DS3.h>
#include <Arduino_JSON.h>

const int grf8 = A1;
const int grf7 = A0;
const int grf6 = A5;
const int grf5 = A6;
const int grf4 = A7;
const int grf3 = A2;
const int grf2 = A3;
const int grf1 = A4;

int gfrValue8 = 0;
int gfrValue7 = 0;
int gfrValue6 = 0;
int gfrValue5 = 0;
int gfrValue4 = 0;
int gfrValue3 = 0;
int gfrValue2 = 0;
int gfrValue1 = 0;

void setup() {
  Serial.begin(115200);
  while (!Serial)
    ;

  if (!IMU.begin()) {
    Serial.println("Failed to initialize IMU!");

    while (1)
      ;
  }

  IMU.accelerationSampleRate();
  IMU.gyroscopeSampleRate();
}

void loop() {
  float x, y, z;
  float Gx, Gy, Gz;

  JSONVar myObject;

  myObject["vers"] = 1;

  gfrValue8 = analogRead(grf8);
  gfrValue7 = analogRead(grf7);
  gfrValue6 = analogRead(grf6);
  gfrValue5 = analogRead(grf5);
  gfrValue4 = analogRead(grf4);
  gfrValue3 = analogRead(grf3);
  gfrValue2 = analogRead(grf2);
  gfrValue1 = analogRead(grf1);

  myObject["gfrValue8"] = gfrValue8;
  myObject["gfrValue7"] = gfrValue7;
  myObject["gfrValue6"] = gfrValue6;
  myObject["gfrValue5"] = gfrValue5;
  myObject["gfrValue4"] = gfrValue4;
  myObject["gfrValue3"] = gfrValue3;
  myObject["gfrValue2"] = gfrValue2;
  myObject["gfrValue1"] = gfrValue1;

  if (IMU.accelerationAvailable()) {
    IMU.readAcceleration(x, y, z);

    myObject["x"] = x;
    myObject["y"] = y;
    myObject["z"] = z;
  }

  if (IMU.gyroscopeAvailable()) {
    IMU.readGyroscope(Gx, Gy, Gz);

    myObject["Gx"] = Gx;
    myObject["Gy"] = Gy;
    myObject["Gz"] = Gz;
  }

  String jsonString = JSON.stringify(myObject);

  Serial.println(jsonString);

  delay(10);
}

```

facchinm, 2023-07-31:
@vongomben the issue is about the "funny" I2C sharing on Nano pinout ( SDA/SCL are shorted with A4/A5). When `analogRead(A4)` is called, the I2C stops being functional  and this locks the sketch when `IMU.accelerationAvailable()` is then called.
If the I2C accelerometer was externally connected, it would be feasible to fix this by using another SERCOM for I2C (https://learn.adafruit.com/using-atsamd21-sercom-to-add-more-spi-i2c-serial-ports/creating-a-new-wire) . Unfortunately, since the connection is internal, there's no chance to use the onboard LSM6DS3 and A4/A5 at the same time :disappointed:

vongomben, 2023-09-08:
Thanks @facchinm 

This is a really precious information

vongomben, 2023-10-31:
I could bypass this by using a demux
https://www.youtube.com/watch?v=Dco6jo9xgAo