vongomben/smart-foot-brace #15: Sole 3d Design (TPU + PLA)?

vongomben, 2023-10-10:
The [BOM is here](https://docs.google.com/spreadsheets/d/e/2PACX-1vR7YMZS-MX4lgYpCrwV-KHx-E0jnsvpv04R4XSrfAN4V9t7ghrDI92tuOamuOgdfb9chOnYX6ybkv2q/pubhtml?gid=705070172&single=true)

As of now, is based on these parts: 

|ref|description                               |label  |Manufacturer|MPN               |DPN               |link                                                                                          |
|---|------------------------------------------|-------|------------|------------------|------------------|----------------------------------------------------------------------------------------------|
|1  |Sensore Soletta                           |ZNX-01 |Qiter       |Qiter4hr835o19f   |B08CXZ6QWM        |https://www.amazon.it/intelligente-pressione-Accessorio-tappetino-sensibile/dp/B08CXZ6QWM     |
|2  |MCU                                       |ESP32  |AZDelivery  |B071P98VTG        |B071P98VTG        |https://www.amazon.it/AZDelivery-NodeMCU-Development-Arduino-gratuito/dp/B071P98VTG/          |
|3  | 3.7V 1000mAh Battery                     |Battery|AKZYTUE     |A202106326        |A202106326        |https://www.amazon.it/AKZYTUE-Batteria-ricaricabile-polimeri-connettore/dp/B0983K9BMK/        |
|4  |TP4056 USB C Tipo C Modulo di ricarica (3)|Charger|RED DAMORE  |B0BJQPN3RG        |B0BJQPN3RG        |https://www.amazon.it/RED-DAMORE-ricarica-batteria-protezione/dp/B0BJQPN3RG/                  |
|~5~  |~DC-DC Buck Converter (6)~                  |Buck   |DUBEUYEW    |B0B4SYBGQL        |B0B4SYBGQL        |https://www.amazon.it/DUBEUYEW-Converter-Modulo-alimentazione-regolabile/dp/B0B4SYBGQL        |
|6  |Interruttore a Scorrimento                |SWITCH |RUNCCI-YUN  |B09TVDZ8P2        |B09TVDZ8P2        |https://www.amazon.it/RUNCCI-YUN-Interruttore-Scorrimento-Posizione-Elettronici/dp/B09TVDZ8P2/|
|7  |FILAMENT                                  |3D     |SUNLU       |‎SLUK-TPU-WT-0.5KG|‎SLUK-TPU-WT-0.5KG|https://www.amazon.it/SUNLU-Filament-Dimensional-Accuracy-Printing/dp/B07R6PMXSP              |
|8  |IMU LSM9Ds1                               |IMU    |Sparkfun    |SEN-13284         |SEN-13284         |https://www.sparkfun.com/products/retired/13284                                               |
|9  |MODULO DC-DC BOOST CONVERTER 2 AMPERE     |Buck   |Pinto       |056.044.3838      |056.044.3838      |                                                                                              |

Here is my idea of the fittings (part numer 9 is in the place of 5, which is not used anymore:

[image]

vongomben, 2023-10-10:
ZNX-01
[image]

vongomben, 2023-10-13:
Review 13 Ottobre. 
grandissimo lavoro di @abenfante 
Problemi alla fine.
[image]
[image]
[image]
[image]
[image]

abenfante, 2023-10-13:
Grazie, tutto chiaro, la canaletta del punto B serve che arrivi allo MCU, giusto?

vongomben, 2023-10-13:
La canaletta deve raggiungere la mcu, e la 4 la farei più larga in entrambe le direzioni

vongomben, 2023-10-17:
Review 17 Ottobre: 
tutto funge, ma dobbiamo shiftware l'esp32 ancora più a sinistra, permettendogli di stare un poì più verso il basso (mi riferisco all'immagine) in modo da peremettere alla presa della suola di rimanere dreitta entrando sotto, e non leggermente ondulata. 

nel punto 1 ho dovuto tagliare qualche millimetro su entrambi i lati, altrimenti la presa USB C non entrava
nel punto  2 shiftando di approx 15mm a sinistra il vano della presa per programmare ESP32. Come dicevo poco più sopra un po' di spazio per l'antenna permetterbbe di abbassare l'ingresso della suola sensoristica. 

LAST AND MORE IMPORTANT: mettere tutto allo stesso livello! ci serve la stessa profondità sul modulo ESP32, la stessa della batteria per intenderci. 

e ci serve il coperchio (opzionale, perché é sicuramente instruttivo il vedere sotto la suola cosa c'é)

se metto in stampa per domani mattina é tutto pronto

Ottimo lavoro @abenfante  

[image]

vongomben, 2023-10-26:
Il prodotto é solido e resiliente. Abbiamo avuto qualche problema nella comuicazione WIFI, molto meglio investire un po' di tempo per creare una micro app ble. 
Unico problema é stato l'ingresso della suola. Ho messo degli elementi fucsia per evidenziarlo. Servirebbe dare più spazio in modo che la superficie trasparente possa aderire facilmente al pavimento e non al bordo 
[image]