vongomben/smart-foot-brace #19: Meeting with Davide Gomba

aam19024, 2024-02-27:
Today our Senior Design Project group met with Mr. Davide Gomba to discuss his work with the ZNX-01 sensor and his projects. The meeting was extremely helpful, especially with discussing within the context of a biomedical sensor. 
Some key points:
- an IMU should be used to account for the balance of an individual
- Wi-Fi signalling may not be possible due to 100ms delay
- Mr. Gomba already has a Bluetooth signalling protocol set up
- nano EVERY is functional as an processor if no additional components are added
- the array of data most useful to us in under the BLE folder

vongomben, 2024-03-06:
I didnt' use the NANO Every in this project because of the multitples analog pins we need in the project. more [here](https://github.com/vongomben/smart-foot-brace/issues/10)

vongomben, 2024-03-06:
most of the ble related things are [here](https://github.com/vongomben/smart-foot-brace/tree/main/ble)